helium/router #804: Support AS923-1 in Pakistan

mikev, 2022-06-23:
Currently the Helium network uses IN865 in Pakistan.  Helium has a customer contact in Pakistan and the customer has determined that IN865 is not legal within Pakistan.  This is primarily because the 800 Mhz band is not intended for IoT in Pakistan, but also the 800 Mhz band in Pakistan is limited to 10 mW but IN865 uses 1000 mW.  The customer has shared regulatory documents which confirm these facts.

Further regulator details:
https://github.com/helium/miner/issues/1741

After a discussion with other Helium engineers we've reached consensus that AS923-1 is the correct plan.  AS923-1 is also listed in the LoRa Regional Spec.

mikev, 2022-08-16:
AS923-1 deployed

closed 2022-08-16 by mikev

iwebconsultants, 2022-08-18:
Hi Team, 

Can I please request to review, as the frequency page is still showing Pakistan in IN865:

https://docs.helium.com/lorawan-on-helium/frequency-plans/

Are you able to assist please ?

Regards,

Azam

waveform06, 2022-11-18:
https://docs.helium.com/lorawan-on-helium/frequency-plans/ has been updated