helium/router #817: Missing Uplinks

RTK221B, 2022-07-08:


jdgemm, 2022-07-20:
Let's see if the new miner update has had a positive impact on their situation like we've seen with other users: https://engineering.helium.com/2022/07/14/miner-hotspot-release.html

Net-Metrics, 2022-08-14:
I'm still seeing this on some devices, others behave as expected. Don't know if it could be related but thought I'd post just in case it points towards something.

My devices with missing uplinks are the same devices that have the LinkADRReq/LinkADRAns Looping issue 

https://github.com/helium/router/issues/791

On a device that behaves as expected I see the following
[image]

This device successfully transmits almost all of it's uplinks, you can see 1 missed uplink at the start of this, every other one was received
[image]
 

Contrast this with the device that has issues
[image]

Received uplinks, this sensor missed 15 uplinks over the same period
[image]

closed 2022-09-07 by jdgemm