helium/router #826: Some nodes are yet to activate on a new console after following instructions

buzzware, 2022-07-18:
**Describe the bug**
I had about 8 devices running fine on console.helium.com, then I was given access to the dedicated console, and followed the instructions to disable, export and import my devices to the new console.
The problem is that some, but not all devices the devices are receiving data on the new console.
Some devices of the same brand and model (but different batch and were onboarded months before the ones that failed) rejoined and continued on the new console without a problem.

**To Reproduce**
1. Setup devices on console.helium.com
2. export devices, including option to disable them
3. import on another console

**Expected behavior**
1. The old console should no longer receive data from the devices and no longer show updates in the "Last Connected" column on console
2. The new console should show the devices having rejoined and data flowing in the "Last Connected" column.
3. If the device needs to rejoin, I think (my knowledge is limited here) confirmed uplinks should not be ACKed or perhaps there is some other protocol signal to cause the device to rejoin. Otherwise, perhaps a rejoin is no longer necessary.

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
 - Console Device ID
 - Device dev eui / app eui
 - How often is device sending data
 - Region
 - Hotspots (names) used to transfer data

Transferred ok :
https://console-vip.helium.com/devices/8b183398-b8a9-4356-beba-ee8cc80e2d2f    *3
https://console-vip.helium.com/devices/c31a7a5a-ce77-41d9-9644-6a6ec7229f55
https://console-vip.helium.com/devices/2ce6c3f8-f226-465a-9381-a335c3613711      *1

Unsuccessful
https://console-vip.helium.com/devices/e2c0ff1c-e38c-49aa-9760-7dc10b2ed115
https://console-vip.helium.com/devices/952f4ec8-6578-4f3e-a913-461362ee1b6f      *2
https://console-vip.helium.com/devices/a0ebc381-6246-491d-a4e5-8e2737725f6a
https://console-vip.helium.com/devices/784c0f3d-54cd-4b14-bc5b-24985a8b7c69

*1 this one is at base on a hill, and perhaps has better reception. The unsuccesful ones are installed at ground level in a suburb with some Helium coverage, at least enough for the old console

*2 this one was deleted from the old console. The other 3 unsuccesfull are still registered on the old console and receiving data

*3 stopped receiving data on the old console at 10AM this morning, and has done 161 uplinks today all without intervention, so has made the transition. Does this mean they will all make the transition with enough time ?

*NOTE: do not post your device `app_key`*

**Additional context**
Add any other context about the problem here.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.

buzzware, 2022-07-19:
Should I delete the old organisation?

buzzware, 2022-07-22:
4 days later, 3 devices are still receiving (within 5 minutes ago) on the old console according to "Last Connected" even though disabled, and have not connected or uplinked at all on the new console.

jdgemm, 2022-07-22:
Sorry, but usually deactivating the device on old Console works. In this case, the team's recommendation is to delete the devices from the old Console and rejoin on new.

buzzware, 2022-07-23:
@jdgemm Deleting after importing didn't work for *2.
They are all deployed in the field an hour away, and we are going there today, so we should be able to resolve it manually.
But I still think there is an unnecessary fault here. If the network refused to ACK the confirmed uplink, it would probably rejoin by itself.

You might say the default console is just for demo purposes, but :

1) it wasn't when these devices were deployed - and there was no 10 device limit
2) the current sales funnel begins with customers easily onboarding devices to the default console. IoT devices are regularly deployed in hard to reach places. There is no warning that you will have to transition and reboot your devices if/when you choose to adopt Helium for real. Not to mention the existence of the dedicated console and the negotiation required with the sales people. Someone needs to straighten out the new customer experience.

buzzware, 2022-08-01:
This one has been lost https://console-vip.helium.com/devices/a0ebc381-6246-491d-a4e5-8e2737725f6a

The other 3 above were still stuck on the old console, so on Friday we stopped the devices, changed the last digit of the DEVEUI to 1 and then asked them to rejoin.
The old console shows they were being received up until Friday.
Now 4 days later, they have not rejoined.
I was hoping that changing the DEVEUI would cause the entire network to treat it as an entirely new device and rejoin on the new console.

Next I guess we will have to swap over and retrieve the devices for closer analysis.

buzzware, 2022-08-31:
After 4 days the devices did eventually join on the new console and work fine. There are separate reasons why it took so long to join - the point is that moving from one console to the other currently requires changing the DEVEUI, not merely importing, deleting etc. This is a poor customer experience, when it is already painful and unnecessary from a customer point of view to change consoles.

closed 2022-09-07 by jdgemm