helium/router #827: Devices communication stops randomly after last release deployment

disk91, 2022-07-18:
**Describe the bug**
Devices communication are not proceeded anymore by router but correctly transmitted by hotspots

The communication stops and is never back. Device that join will basically have no later communication when the first transmission has been lost. 

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
seen on console.helium.com and console.helium-iot.eu at least

**Additional context**
Start appearing right after router / console upgrade to 2.2.13, but the communication did not stopped for all and not all at the same time

r4wk, 2022-07-18:
I'm only seeing hot spots that are already running gateway-rs doing data transfers. Hot spots running gateway-rs are able to hear my uplinks and also downlink properly. Any hot spot NOT running gateway-rs do not hear my uplinks. 

I know other people are seeing different issues, but this is my experience and is very easy to see in the console and a coverage map.

disk91, 2022-07-18:
I also have similar problem with gateway-rs hotspots

disk91, 2022-07-18:
in router logs (not sure related)
```
[2022-07-18 16:04:44.324] <0.1791.0> [info] [router_metrics:handle_info:{256,5}] running metrics
[2022-07-18 16:04:50.524] <0.16056.28> [error] [] gen_statem <0.16056.28> in state connected terminated with reason: {etimedout,state_functions}
[2022-07-18 16:04:50.525] <0.16056.28> [error] [] CRASH REPORT Process <0.16056.28> with 0 neighbours exited with reason: etimedout in gen_statem:loop_state_callback_result/11 line 1397
[2022-07-18 16:04:50.525] <0.1497.0> [error] [] Supervisor 'grpcbox_pool_0.0.0.0_8080' had child {grpcbox_acceptor,{{84,153,98,177},34558},
                  {{172,23,0,4},8080},
                  #Ref<0.212979515.4156030977.42330>} started with {grpcbox_acceptor,acceptor_init,undefined} at <0.16056.28> exit with reason etimedout in context child_terminated
[2022-07-18 16:04:50.525] <0.5467.29> [error] [] gen_statem <0.5467.29> in state open terminated with reason: {etimedout,state_functions}
[2022-07-18 16:04:50.526] <0.5467.29> [error] [] CRASH REPORT Process <0.5467.29> with 0 neighbours exited with reason: etimedout in gen_statem:loop_receive/3 line 1078
[2022-07-18 16:04:50.526] <0.5268.29> [error] [] gen_statem <0.5268.29> in state half_closed_remote terminated with reason: {etimedout,state_functions}
[2022-07-18 16:04:50.526] <0.5268.29> [error] [] CRASH REPORT Process <0.5268.29> with 0 neighbours exited with reason: etimedout in gen_statem:loop_receive/3 line 1078
[2022-07-18 16:04:50.526] <0.4302.29> [error] [] gen_statem <0.4302.29> in state half_closed_remote terminated with reason: {etimedout,state_functions}
[2022-07-18 16:04:50.526] <0.4302.29> [error] [] CRASH REPORT Process <0.4302.29> with 0 neighbours exited with reason: etimedout in gen_statem:loop_receive/3 line 1078
```

disk91, 2022-07-18:
Other information : one of the state channel close submit has been in error

ke6jjj, 2022-07-18:
There is a confirmed issue with the miner release `2022.07.14.0` and how it behaves when the `gateway_and_mux_enable` application environment variable is set to `false` (which is the default value). In this mode the Erlang miner will be in charge of data packet routing, but due to an unfortunate bug, it will crash itself when handling any post-join data packet -- any packet which has a LoRaWAN Device Address present.

The fix is to either:
* Enable the `gateway_and_mux_enable` variable in the `sys.config` file, which will put the Rust `gateway-rs` in charge of data packets, or
* Build and deploy a new version of the miner, once <https://github.com/helium/blockchain-core/pull/1429> has been merged.

ke6jjj, 2022-07-18:
@disk91 I'm curious if the issue I just described is the most likely source of what you're seeing. If you have the time, I'd like to confirm whether the mux is enabled or not on the Hotspots in question. Obviously, this will depend on the `sys.config` file, and thus, the maker of the Hotspot and their release schedule. If you know these, please post.

r4wk, 2022-07-18:
> @disk91 I'm curious if the issue I just described is the most likely source of what you're seeing. If you have the time, I'd like to confirm whether the mux is enabled or not on the Hotspots in [...]

In my case this is most certainly the issue. I can tell because the only hot spots hearing me are the ones who already have MUX=true.

ke6jjj, 2022-07-18:
> I got it from a gateway-rs, is there a specific setting also on gateway-rs ? in my case, it sometime works and sometime not work. easy to reproduce if needed

No, the issue I'm describing is specific to the Erlang miner. If you're just running `gateway-rs` then I don't have a theory (yet).

ke6jjj, 2022-07-18:
@disk91 You're absolutely right that we haven't definitely narrowed down the issue you're experiencing here, for sure. But it's possible that the upgrade you performed to 2.2.14 just coincidentally happened to be at roughly the same time when some of the Hotspot makers started turning on the mux in their software.

disk91, 2022-07-18:
> @disk91 You're absolutely right that we haven't definitely narrowed down the issue you're experiencing here, for sure. But it's possible that the upgrade you performed to 2.2.14 just coincidentally [...]

I agree for the timing but most of the unlocked devices were covered by RAK and DIY hotspots not yet updated.
But my initial investigation is about a single gateway-rs located in France but working in US915 for debugging purpose, so I'm quite confident that the traffic is not going through a different path.

leogaggl, 2022-07-19:
We were experiencing the same issue with Milesight gateways against multiple routers. None had transferred data packets since approx Friday (both in Europe as well as in Australia).

We were able to work around this by setting `gateway_and_mux_enable=true` as Milesight has not yet released an update overriding the default and disabling chain sync.

What I don't get is if chain sync is still enabled on the gateway should this not be available to the miner via the fully synced local blockchain?

closed 2022-09-07 by jdgemm