helium/router #851: The devices are unable to send uplink after contacting with one exact hotspot

rdrill, 2022-08-11:
**Describe the bug**
The uplinks are coming stable until one particular hotspot (Cool Satin Giraffe) reported **bad_message_type** error. After that device is trying to send uplinks without any success, the other gateways ignore them. For me, it looks like the device got some ban in a network.  Then, in some time (10 min - 3+ hours) the problem vanished and then reappear after the same hotspot would catch an uplink.
After the "releasing" console shows a lot of late uplinks from other gateways.
The issue has been reproduced on 3  different devices.

**To Reproduce**
Looks like this is something local and related to one particular hotspot

**Expected behavior**
The device must not be blocked by the network because of one glitchy hotspot. The uplink must be received successfully, as long as other hotspots receive the uplinks from the same device successfully, the hardware debugger message showed that the LoRaWAN packet structure is the same and the message type == 4 in every sending case.

**Device Info (please complete the following information):**
 - Environment (Production, Staging): Stagging
 - Console Device ID: 52897f9f-16c5-4a7e-aa26-3ac5ec67caf1
 - Device dev eui / app eui: 6081F99B5FD93742 / 6081F9AE8058F59C
 - How often is the device sending data: the issue exposed with both 2m data interval (debug mode) and 15 minutes data interval for 5-7 bytes payload
 - Region: EU868
 - Hotspots (names) used to transfer data: The problems are only with Cool Satin Giraffe, no problems with other ones (Witty Shamrock Pike, Spare Smoke Woodpecker, and 5+ others)

**Screenshots**
[image]
[image]
[image]

vkynchev, 2022-09-03:
The issue looks like #855 which I’ve opened ~2 weeks after this issue. There’s a possible solution which I’ve proposed there for the devs to take a look at…

macpie, 2022-09-13:
Seems like a duplicate of #855 . Will fix in #855

closed 2022-09-13 by macpie