helium/router #854: Need Help: p2p  connection issue

YoSmartInc, 2022-08-27:
**Describe the bug**
helium_router failed to connect to p2p node.

**To Reproduce**
Host my own console in Amazon EC2  according to the Guide, But failed to connect to the blockchain.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Device Info (please complete the following information):**
 - Environment (Production, Staging) 
 - Console Device ID
 - Device dev eui / app eui
 - How often is device sending data
 - Region
 - Hotspots (names) used to transfer data

*NOTE: do not post your device `app_key`*
My OUI: 43
My router address: 118t8HZ4dSnwC5C9hxyM4HQQUKMczo2zoe86ten5aYBHW1wnrY5

**Additional context**
Here is the log of helium_router -> router.log:
[2022-08-27 10:29:53.966] <0.1751.0> [info] [router_console_dc_tracker:init:{156,5}] router_console_dc_tracker init with #{}
[2022-08-27 10:29:53.967] <0.1753.0> [info] [router_v8:init:{57,5}] router_v8 init with #{}
[2022-08-27 10:29:53.971] <0.1757.0> [info] [router_device_devaddr:init:{123,5}] router_device_devaddr init with #{}
[2022-08-27 10:29:53.974] <0.1758.0> [info] [router_xor_filter_worker:init:{175,5}] router_xor_filter_worker init with #{}
[2022-08-27 10:29:53.974] <0.1749.0> [warning] [router_console_ws_worker:handle_info:{283,5}] rcvd unknown info msg: {ws_message,<<"device:all">>,<<"phx_reply">>,#{<<"response">> => #{},<<"status">> => 
<<"ok">>}}, {state,<0.1750.0>,<<"ws://helium_console:4000/socket/router/websocket">>,#Ref<0.140648651.425590795.83649>,#Ref<0.140648651.425590795.83651>}
[2022-08-27 10:29:53.974] <0.1749.0> [warning] [router_console_ws_worker:handle_info:{283,5}] rcvd unknown info msg: {ws_message,<<"organization:all">>,<<"phx_reply">>,#{<<"response">> => #{},<<"status"
>> => <<"ok">>}}, {state,<0.1750.0>,<<"ws://helium_console:4000/socket/router/websocket">>,#Ref<0.140648651.425590795.83649>,#Ref<0.140648651.425590795.83651>}
[2022-08-27 10:29:53.974] <0.1749.0> [warning] [router_console_ws_worker:handle_info:{283,5}] rcvd unknown info msg: {ws_message,<<"label:all">>,<<"phx_reply">>,#{<<"response">> => #{},<<"status">> => <
<"ok">>}}, {state,<0.1750.0>,<<"ws://helium_console:4000/socket/router/websocket">>,#Ref<0.140648651.425590795.83649>,#Ref<0.140648651.425590795.83651>}
[2022-08-27 10:29:53.974] <0.1749.0> [warning] [router_console_ws_worker:handle_info:{283,5}] rcvd unknown info msg: {ws_message,<<"organization:all">>,<<"phx_reply">>,#{<<"response">> => #{},<<"status"
>> => <<"ok">>}}, {state,<0.1750.0>,<<"ws://helium_console:4000/socket/router/websocket">>,#Ref<0.140648651.425590795.83649>,#Ref<0.140648651.425590795.83651>}
[2022-08-27 10:29:54.435] <0.1714.0> [warning] [blockchain_state_channels_server:load_state_channels:{701,5}] not in db sc ids: []
[2022-08-27 10:29:54.435] <0.1714.0> [warning] [blockchain_state_channels_server:load_state_channels:{709,5}] presumably closed sc ids: []
[2022-08-27 10:29:54.436] <0.1714.0> [warning] [blockchain_state_channels_server:get_new_active:{469,13}] don't have any state channel left unused
[2022-08-27 10:29:54.492] <0.1758.0> [error] [router_xor_filter_worker:get_filters:{666,13}] failed to find routing for OUI: 43 not_found
[2022-08-27 10:29:54.492] <0.1758.0> [error] [router_xor_filter_worker:should_update_filters:{921,13}] failed to find routing for OUI: 43 not_found
[2022-08-27 10:29:54.492] <0.1758.0> [info] [router_xor_filter_worker:handle_info:{398,13}] filters are still up to date
[2022-08-27 10:29:54.721] <0.1650.0> [warning] [libp2p_nat:maybe_apply_nat_map:{137,37}] invalid nat map: {[],[]} {[],[]} {error,{badmatch,{error,einval}}}
[2022-08-27 10:29:54.811] <0.1650.0> [warning] [libp2p_nat:maybe_apply_nat_map:{137,37}] invalid nat map: {[],[]} {[],[]} {error,{badmatch,{error,einval}}}
[2022-08-27 10:29:55.017] <0.1650.0> [warning] [libp2p_nat:maybe_apply_nat_map:{137,37}] invalid nat map: {[],[]} {[],[]} {error,{badmatch,{error,einval}}}
[2022-08-27 10:30:04.432] <0.1715.0> [info] [router_metrics:handle_info:{256,5}] running metrics
[2022-08-27 10:30:08.963] <0.1745.0> [info] [router_sc_worker:maybe_start_state_channel:{354,13}] [active: 0] [overspent: 0] [getting close: 0] [opened: 0] [in flight 0] [max: 25] all active, opening mo
re
[2022-08-27 10:30:08.963] <0.1745.0> [info] [router_sc_worker:open_next_state_channel:{415,17}] failed to get a good expiration no_opened_sc
[2022-08-27 10:30:08.967] <0.1745.0> [info] [router_sc_worker:create_and_send_sc_open_txn:{456,5}] Opening state channel for router: "118t8HZ4dSnwC5C9hxyM4HQQUKMczo2zoe86ten5aYBHW1wnrY5", oui: 43, nonce
: 1, id: <<79,223,228,25,159,12,56,5,47,53,214,251,158,111,225,234,45,113,231,137,0,149,121,100,84,193,65,79,61,177,78,90>>
[2022-08-27 10:30:09.927] <0.1799.0> [warning] [libp2p_transport_proxy:connect_rcv:{116,9}] timeout_proxy_session proxying to "/p2p/11BLPytHFxQYnjE7wEooEUzZrH1KXahHN7Vd3i4Bn6fubzoiS9K"
[2022-08-27 10:30:09.928] <0.1657.0> [warning] [libp2p_group_worker:connecting:{238,5}] Failed to connect to "/p2p/112EGCZWqMbDZNR7jL4LfQ4zrDxPNDN5Vp89Jd3BQG6f1j8x8WvG": [{"/p2p/11BLPytHFxQYnjE7wEooEUzZ
rH1KXahHN7Vd3i4Bn6fubzoiS9K/p2p-circuit/p2p/112EGCZWqMbDZNR7jL4LfQ4zrDxPNDN5Vp89Jd3BQG6f1j8x8WvG",timeout_proxy_session}]
[2022-08-27 10:30:11.699] <0.1676.0> [warning] [libp2p_nat_server:delete_mapping:{192,13}] failed to delete port mapping {2154,2154}: no_nat

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.

YoSmartInc, 2022-08-28:
I had hosted another router(address:112iF2fACBMpZXx5KXvQ4XW89LHPDmpRyB5rBGBwspBN9vZbcJe1),  but the same thing happened.

michaeldjeffrey, 2022-08-29:
I don't think we log when connections are successful, only when they fail. 

For the `no_nat` messsages, you may need to fill out this part of the `.env` file.
https://github.com/helium/router/blob/master/.env-template#L4-L8

That's mapping your internal AWS address to the external AWS address for gossip purposes, as far as I understand it.

You can also see how many connections your router has by running:
```sh
docker exec -it helium_router _build/default/rel/router/bin/router peer book --count
```

closed 2022-08-29 by jdgemm