helium/router #855: ACKs are sometimes treated as dropped uplinks (bad_message_type)

vkynchev, 2022-08-27:
**Describe the bug**
Sometimes console treats forwarded ACKs as uplinks and they get marked as "Uplink Dropped" because of bad_message_type. 
Up to that point everything seems okay, but when fcntDn gets to the point of `fcntDn = fcntUp + N` and `N > 1` issues begin to happen. This happens primarily because the end device didn't receive the ACK and tries to re-transmit the same uplink (examples of `uplink/downlink` are `4/4`, `4/5` and `4/6`  in the screenshot). On the next transmission (`5/8`) the ACK gets treated as an uplink by another gateway and gets forwarded to console, which sets 8 as current fcntUp. On next transmission from the end-device (has to be `6/9`) the packet gets discarded by console, because it expects to have fcntUp >= 8 (because of the ACK issue). This blocks device uplinks until the device decides to abandon this packet and increment the fcntUp. This might take from couple minutes to couple days (or even more, depending on how frequent the device is transmitting and how big is the gap `N` between console's fcntUp and fcntDn).

**To Reproduce**
Configure an end-device to send confirmed uplinks with SF10 or SF11 in an area with a lot of gateways/hotspots (a city centre for example). After a while (< 20 packets transmitted, before ADR kicks in) you should see ACKs forwarded by some gateway to the console. This might or might not happen on other locations, depending on what software the gateways are using. I'm having issue with `brilliant-cyan-mammoth` forwarding the ACKs. 

**Expected behavior**
Console shouldn't treat `bad_message_type` uplinks as real uplinks. fcntUp should be set to the last successful uplink. This way even if the packet is real (not false ACK "uplink") it'll get sorted out.

**Device Info (please complete the following information):**
 - Environment (Production, Staging): Production
 - Console Device ID: ddded7d0-1837-4a51-8c58-d704e93fe7da
 - Device dev eui / app eui: 500291362198FEEC / 6081F9C80023ADC6
 - How often is device sending data: every 60s for testing purposes
 - Region: EU868
 - Hotspots (names) used to transfer data: `brilliant-cyan-mammoth` (bad one), `zany-gingham-panther`, `icy-admiral-quail`, `wonderful-tawny-albatross`, `sneaky-saffron-lizard` and a couple more. Listed are the main hotspots that I've seen the most packets forwarded.

*NOTE: do not post your device `app_key`*

**Additional context**
Add any other context about the problem here.

**Screenshots**
[image]

vkynchev, 2022-09-03:
On my other location I am experiencing issues with this hotspot `Petite Misty Raccoon`. Also it looks like it's still getting paid for the invalid packets (I even don't see that hotspot in Coverage page in Console...). I'm pretty confidend to say that these are my packets, because 1st - no one is ever gonna blast that much messages (most sensors send data at least 15 min apart) and 2nd before I deploy the test fleet on that location not even a single hotspot was getting data transfer rewards. I'm not sure if I'm paying for them or the rewards are sent from a reward pool or something, but it definitely looks odd...

vkynchev, 2022-09-03:
Here are some more screenshots of the issue. It's even happening right now (11:40pm local time, to compare to the screenshots)...

[image]
[image]
[image]
[image]

vkynchev, 2022-09-03:
I did a bit of digging in the source code (excuse me if I'm wrong, but not an erlang dev here) and here's what I found.

[This change](https://github.com/helium/router/pull/846/files#diff-bf775c7152560673e6f20eeff171132bc355b1790c8d753e2b2a8b939f4f8d08R1495) brought that issue to the table, but it's not the root cause. It just changed the check from != to =< for late packets.

What I suspect happens is [here](https://github.com/helium/router/blob/master/src/device/router_device_worker.erl#L722-L723) and [here](https://github.com/helium/router/blob/master/src/device/router_device_worker.erl#L744) the device last fcnt is changed and saved before validating the packet. Unfortunately I don't have any testnet hotspots and HNT tokens so I won't be able to test anything.

macpie, 2022-09-13:
@vkynchev Thanks I really appreciate the detailed explanation. 

I think the best we can do here is to check the message type prior to sending it to the device worker so that it never has to receive a forwarded downlink as an uplink.

macpie, 2022-09-13:
Fixed with https://github.com/helium/router/pull/860

closed 2022-09-13 by macpie

vkynchev, 2022-09-16:
Awesome. I confirm that the fix works as expected. Thank you!

PR #860 (2022-09-13): Match only uplinks to avoid sending forwarded downlinks to device worker
Fix #855