helium/router #856: Problem with different Frequency for up and downlinks 868 vs 433

KambaComms, 2022-08-29:
I am using a Heltec Hotspot that has been set to EU868 (The correct frequency for Zimbabwe where it is installed.) It was working fine until about 4 months ago, after which a Console "upgrade" broke it. Since then it appears that the Dragino LGT-92 that I am using to send tracking information to the hotspot is sending on EU868 but is receiving on 433. This is what I can see from the Console. I can't see how I can change this.

**To Reproduce**
Use a Dragino LGT-92 and a Heltec Helium Hotspot in Zimbabwe, All set up to use EU868 regional settings.

**Expected behavior**
I expected both up and downliks to show EU868 settings from the console.

**Device Info (please complete the following information):**
 - Production Environment
 - Console Device ID  91b0e325-2c39-4561-8725-c40977936ef4
 - Device dev eui / app eui   A840411FA182FF58
 - How often is device sending data  Every 2 mins
 - Region EU868 Zimbabwe
 - Hotspots (names) used to transfer data old-bone-swift

*NOTE: do not post your device `app_key`*

**Additional context**
Add any other context about the problem here.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.

[image]
[image]

michaeldjeffrey, 2022-08-29:
Something weird is going on, you'll have to contact Heltec.

The gateway is obviously listening in the 868 range. 
But the Region being sent in the state_channel_packet is `EU433`. 

The region in the packet comes from and is signed by the gateway.

closed 2022-08-29 by jdgemm

KambaComms, 2022-08-30:
[image]

Helium has the incorrect frequency set in its channel plan for Zimbabwe. How can this be changed.