helium/router #857: Helium has the incorrect frequency set for Zimbabwe. It should be EU868 not EU433

KambaComms, 2022-08-30:
**Describe the bug**
The console is trying to force EU433 on hotspots in Zimbabwe.

**To Reproduce**
Register a hotspot in Zimbabwe with the correct frequency EU868 and the console will attempt to force EU433

**Expected behavior**
The console should set EU868 for Zimbabwe so that manufacturers do not mis-configure the firmware to use anything else.

Production Environment
Console Device ID 91b0e325-2c39-4561-8725-c40977936ef4
Device dev eui / app eui A840411FA182FF58
How often is device sending data Every 2 mins
Region EU868 Zimbabwe
Hotspots (names) used to transfer data old-bone-swift

*NOTE: do not post your device `app_key`*

**Additional context**
Add any other context about the problem here.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.
[image]

[image]

[image]

mikev, 2022-08-30:
I've confirmed that per the LoRa Alliance Zimbabwe should be EU868.  We will switch from HPlans for Zimbabwe from EU433 to EU868.
https://github.com/dewi-alliance/hplans/blob/main/regions.csv

mikev, 2022-09-09:
Frequency issues is now addressed.  Helium will switch from EU433 to EU868.
https://github.com/helium/lorawan-h3/commits/main

closed 2022-09-09 by mikev

KambaComms, 2022-10-23:
Is there any progress on this? I still have 433 coming out of my Heltec Hotspot.