helium/router #871: Device migration script from Router/Console to Chripstack

macpie, 2022-10-31:


Dangerousfish, 2022-11-01:
Hope this helps:

GET https://console.helium.com/api/v1/devices
https://docs.helium.com/api/console/#devices-list

Chirpstack:
POST http://<_ChirpStackHost_>:8080/api/devices
http://35.160.9.254:8090/#/DeviceService/DeviceService_Create

https://www.chirpstack.io/docs/chirpstack/api/grpc.html

macpie, 2022-11-01:
> Hope this helps:
> 
> GET https://console.helium.com/api/v1/devices https://docs.helium.com/api/console/#devices-list
> 
> Chirpstack: POST http://<_ChirpStackHost_>:8080/api/devices http://<_ChirpstackHost_>:8090/#/DeviceService/DeviceService_Create
> 
> https://www.chirpstack.io/docs/chirpstack/api/grpc.html

Thank you any help is always welcomed!

macpie, 2022-11-10:
Device GRPC def https://github.com/chirpstack/chirpstack/blob/master/api/proto/api/device.proto

lytzV, 2022-12-08:
chrpstack migration in english

https://docs.google.com/document/d/1k99_yt41_gV3KVUD9EIYfX_qWdJxl3y6sn-U0jkwcwU/edit?usp=sharing

macpie, 2023-06-22:
Was done by third party

closed 2023-06-22 by macpie