helium/router #881: ADR/FSB2 issue after Dualplan implementation

deeeblack, 2022-11-26:
**Describe the bug**
When ADR is turned on in the firmware of a Dragino device (LHT65N) it requests ADR every 64 uplinks. When the console replies to the device it is telling it to move to FSB2. This then renders the device inoperable as it switches away from the frequencies in use by the new AUS/NZ Dualplan implemented a week or so ago.

**To Reproduce**
Power Down sensor, Power up sensor, Join/Ack completes. The sensors send 64 uplink frames and request ADR. Console replies with a MAC command telling the device to move to FSB2. The sensor then does not have communication with the network.

Even if a sensor has a profile set to not allow ADR, Console still sends it. I have tried this profile function on and off.

**Expected behavior**
Sensor does not move to FSB2 under any circumstances.

**Device Info (please complete the following information):**
 - Environment - **Dev**
 - Console Device ID - **2d64fec5-e338-4300-8e96-5d7970f785b9**
 - Device dev eui / app eui - **A840419B218573D0 / A840410000000100**
 - How often is device sending data - **20minute intervals**
 - Region - **AU915 (Now Dualplan)**
 - Hotspots (names) used to transfer data - **Precise Rusty Mantis (Nebra), cheery pine rabbit (Sensecap)**

**Additional context**
Yes I could turn ADR off, but should I have to?
Yes I could force the device to FSB6, should I have to thou?
I am happy to do the above, but i feel as if this is just s short term fix.

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.
[image]

mikev, 2022-11-26:
ADR is handled in the lorawan library.  The lib is missing a case for AU915_SB5
https://github.com/helium/erlang-lorawan/blob/master/src/lora_chmask.erl

mikev, 2022-11-29:
An issue with the AU915_SB5 chmask is fixed with a lib update.
https://github.com/helium/router/pull/884

mikev, 2022-12-01:
Fix was deployed to the network this morning at 9AM PST.

closed 2022-12-01 by mikev