helium/router #882: AU915 devices not able to OTAA JOIN on 2 channels on so-called "dual-plan"

leogaggl, 2022-11-28:
This has originally been raised by @tonysmith55, @mcauser and @zigon - just creating an issue for this as nobody else has after checking repo issues.

**Describe the bug**
@mvierling to assist with some diagnostics, wonder if you could review and correct my assumption on how the Dual Band code is operating.
1) - "Dual Band" is currently active on the server
2) - The Hotspot (gateway) receive frequencies are configured by global_conf.json and since the receive frequencies AS923-1 and Au915 FSB6 are the same you can use the AS923-1 global config file for Dual Band.  Equally we could use the AU915 FSB6 global file as it would have the same contents.
3) - Therefore you can't tell from just the global_conf.json file if Dual Band is active or not.
4) - The logic to distinquish between a Node on AS923-1 and AU915 is something like the following
    * If a Node sends a Join Request on channels 3-8 then the server assumes the node is  AU915 and respond responds with a Join Accept on the appropriate AU915 channel
    * if the Node sends a Join Request on channels 1 or 2, the server cannot tell if its an AU923 or AU915.  As a result the server assumes its AS923 and responds on the same channel (1 or 2).
    * if it was a AS923 node then the Join Accept is received and node is now part of the network
    * if it is a AU915 node the Join Accept won't be received as the frequency/SF will be wrong and so the AU915 Node does not join and so sends another Join Request on a different channel.  This keeps occurring until it sends a request on channels 3-8 of the FSB6 band.

I also assume the logic above is quite separate from the Tx Power issue. 

**Original message**:
https://discord.com/channels/404106811252408320/1043247093831192587/1046231438351147068
-------------------------------------------------------------------------------------------------------------------
I figured out what the problem is with AU915 joins on AS923 hotspots
If you send a Join Request on AU915 ch40 or 41, the hotspot assumes you're an AS923 device and responds with a JoinAccept on the wrong freq / sf
[image]
If you send a JoinRequest on ch 42-47, it responds as if you're an AU915 device, with 30dbi
Send a JoinRequest on ch 40-41, and the gateway responds as if you're an AS923 device, with 16dbi
The gateway seems to rely on the channel number (freq) to figure out how to respond

Original message:
https://discord.com/channels/404106811252408320/1043247093831192587/1046416955810267207

**To Reproduce**
This behaviour is consistent across multiple devices and gateways

**Expected behavior**
Devices are expected to OTAA JOIN on all channels

mcauser, 2022-11-28:
Would it be helpful if I provide join_request / join_accept ids returned by the console for any of these tests?

leogaggl, 2022-11-29:
There is some further information on this:

The LNS response on these 2 channels is also incorrectly BW125 instead of BW500.

mikev, 2022-12-15:
Behaving as expected.

AU915 Joins can only be handled on Ch 42 - 47 as Ch 40 and 41 are reserved for AS923. A future optimization might handle this differently though and it is technically possible that once a sensor is identified as AU915 all future Joins can use Ch 40 to 47 for Joins. Alternatively the sensor could be labeled as AU915-only in the Console. Per the spec this is valid AU915 behavior.

closed 2022-12-15 by mikev

leogaggl, 2022-12-16:
It would be appreciated if you could point us the where in the spec you see that 2 channels of 8 not responding to JOIN requests on the correct frequency is valid per spec @mikev 

And by "expected" I assume you are referring just to yourself?