helium/router #883: Incorrect TX POWER response on AU15 FSB-6 on so-called "dual-plan"

leogaggl, 2022-11-28:
This has originally been raised by @zigon - just creating an issue for this as nobody else has after checking repo issues.

**Description**

I have a gateway and a critical device that I need reliable transmission from. I have access to the firmware of the device so I can get it to work on any frequency band no problem. However, the device is a long way away from the gateway and can only just pickup join accepts and downlinks when the gateway transmits at 30dbm. From what I can see with FSB6, when the device transmits on the upper 6 channels the LNS correctly interprets it as AU915 FSB and sends the join accept to the gateway on the right frequency at 30dbm. However, the gateways are now on the AS923 radio config and as such the gateway limits the power to 16dbm from what Louis has said. As such there is no way that the device will be able to receiver the join accept.  I imagine this issue will take a long time to get fixed as the gateways need a new radio configuration file to allow 30dbm transmission? Any very rough time estimation for this issue to be fixes? Like days, weeks or months? (edited)

https://discord.com/channels/404106811252408320/1043247093831192587/1044387161501343784

**To Reproduce**
This behavior is consistent and happens with multiple devices.

**Expected behavior**
LNS TX POWER should be 30dBm on all channels

**Additional context**
This has been confirmed by multiple users

mikev, 2022-12-13:
First, there are no known Semtech sensors which transmit at 30dBm.  Semtech Engineering reports to us that:

Semtech sensor chips supports +15 dBm (SX1261), +22 dBm (SX1262) or both (LR1110).
Sensor hardware can implement an external PA to increase it to +30 dBm, but they do not know how often hardware manufacturer are doing this.  Likely this is rare.

Helium Hotspots are currently limited by hardware design to 27dBm.

The AU915 sensor will Tx at 22dBm.  An AU915 FSB2 Helium Hotspot will Rx at 27dBm.

For a DualPlan Hotspot the behavior is the same.
The AU915 FSB6 sensor will Tx at 22dBm.  The 'DualPlan' AU915 FSB6 Hotspot will Rx at 27dBm if it is an AU915 sensor.
The type of sensor is determined by the Join frequency - Ch 40 or 41 determine AS923.  Ch 42 - 47 determine AU915.
Multiple Hotspot owners have confirmed the current behavior is correct.

I hope this clarifies things.

closed 2022-12-13 by mikev

leogaggl, 2022-12-14:
I have checked this with @Zigon who has originally reported that and that statement seems incorrect:

> The 'DualPlan' AU915 FSB6 Hotspot will Rx at 27dBm if it is an AU915 sensor.

The correct Rx response is only on 6 of 8 channels.

@plainsimpledot has agreed to also test this with the sensors set up for testing purposes.

mikev, 2022-12-15:
AU915 Uplinks and Downlinks (@27dBm) are successfully handled on all 8 of 8 channels.

AU915 Joins can only be handled on Ch 42 - 47 as Ch 40 and 41 are reserved for AS923.  A future optimization might handle this differently though and it is technically possible that once a sensor is identified as AU915 all future Joins can use Ch 40 to 47 for Joins.  Per the spec this is valid behavior.