helium/router #892: Malaysia AS923_1B Plan Issue

RTK221B, 2022-12-14:
Please find below the device logs of a device deployed in Malaysia AS923_1B.

Two issues can be seen in the logs:

1. Rx1 Join Response Frequency is wrong - it should mirror Join frequency
2. Rx2 is supposed to be fallback. rx2 reply at 923.2 and 923.4 are not working.

>{
    "category": "join_accept",
    "data": {
      "devaddr": "09040048",
      "fcnt": 0,
      "hotspot": {
        "channel": 0,
        "frequency": 920.4,
        "id": "11q3pZBTp9q5MftEbnrpZNkbNQVogidpVrRgu4h8XvsXvA9KdLf",
        "lat": 5.31729669556136,
        "long": 100.47077108175583,
        "name": "stale-tartan-mantaray",
        "rssi": 16,
        "snr": 0,
        "spreading": "SF10BW125"
      },
      "payload": "IIa3RfwZjYoX5KyMKvhYbfQ=",
      "payload_size": 17,
      "port": 0
    },
    "description": "Join accept from AppEUI: 6081F948DB43D014 DevEUI: 6081F988A3137D32",
    "device_id": "6154a5b1-5407-4fb5-af75-3c8d58cbc0d9",
    "frame_down": null,
    "frame_up": 0,
    "id": "75478aad-d551-4144-9212-c0572a51e879",
    "organization_id": "c03c6dd0-0239-4e1a-b0d2-763ced865c34",
    "reported_at": "1670984939647",
    "router_uuid": "004a72db-806e-49a9-948b-7b44f286dcff",
    "sub_category": "undefined"
  },
  {
    "category": "join_request",
    "data": {
      "dc": {
        "balance": 250,
        "nonce": 1,
        "used": 1
      },
      "devaddr": "09040048",
      "fcnt": 0,
      "hotspot": {
        "channel": 6,
        "frequency": 923.2000122070312,
        "id": "11q3pZBTp9q5MftEbnrpZNkbNQVogidpVrRgu4h8XvsXvA9KdLf",
        "lat": 5.31729669556136,
        "long": 100.47077108175583,
        "name": "stale-tartan-mantaray",
        "rssi": -115,
        "snr": -11.199999809265137,
        "spreading": "SF10BW125"
      },
      "payload": "",
      "payload_size": 0,
      "port": 0,
      "raw_packet": "ABTQQ9tI+YFgMn0To4j5gWB2H66D+Tw="
    },
    "description": "Join request from AppEUI: 6081F948DB43D014 DevEUI: 6081F988A3137D32",
    "device_id": "6154a5b1-5407-4fb5-af75-3c8d58cbc0d9",
    "frame_down": null,
    "frame_up": 0,
    "id": "c73123bd-78ac-4fd7-abb8-d1493e81a928",
    "organization_id": "c03c6dd0-0239-4e1a-b0d2-763ced865c34",
    "reported_at": "1670984937647",
    "router_uuid": "20794d01-00a5-47bf-b6c6-eeb6f07e0405",
    "sub_category": "undefined"
  },

oberchoo, 2023-01-26:
What is next, I am from Malaysia, we have been using Helium network for some projects and suddenly the nodes cannot connect to the network.

leogaggl, 2023-01-26:
I suggest you connect with the Australian community. We have been through very similar issues here, and unless you don't make a lot of noise about it, nothing seems to happen at Helium. It is a totally US-focused operation.

We have a local Discord (https://discord.gg/ZJnUtfNz), and you could possibly join our community calls to put a better spotlight on your issues.