helium/router #924: Unknown Decoder - IoTracker3

thebrighthood, 2023-03-06:
**1. Are you using Helium hosted Console or running your own open source Console?**
Helium Hosted Console

**2. If you are on Helium hosted Console, which url are you currently on?**
https://console.helium.com/devices/56b955a1-1d39-4516-9e5b-908bcf103606

**3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID.**
Org: SlidingLines
DevID: 56b955a1-1d39-4516-9e5b-908bcf103606

**Please describe your issue:**

Unknown decoder in debug screen. Tried to hook things up again from scratch, but it always returns unknown decoder.

I can grab the payload and run it through the **Script Validator and it outputs in the correct format**.
This is the decoder: [https://github.com/ioTracker/decoder-js/blob/master/decoder_ttn.js](url)
The device is an [IoThings IoTracker3.](https://docs.iotracker.eu/decoding/decoder-example/)

I believe this decoder code is OK, because it gives the correct results in the Script Validator.

[image]
[image]

ApolloDS, 2023-10-01:
I have exactly the same problem here. Owning a IOTracker 3 and always get this error.

ApolloDS, 2023-10-02:
Hi @thebrighthood ,
I figured it out, please see here: https://github.com/ioTracker/decoder-js/issues/1
Just add the 'port' variable name to the function and somehow this will work then in the helium console.

Peter