helium/router #977: disable device euis still shows data

ccall48, 2023-07-08:
**Describe the bug**
on disabling a device in fdn console the device euis do not appear to be removed from router route id all the time.

**To Reproduce**
disabled device in fdn console will show activity and in some cases interrupt join/accept if the device has been moved to an open lns. 

**Expected behavior**
confirmed removal of device euis from fdn route id on a disable toggle.

**Device Info (please complete the following information):**
 - Environment (fdn console -> openlns)
 - Console Device ID -> dragino LG92
 - Device dev eui / app eui (A840419B3182234A / A84041F43182234A)
 - How often is device sending data -> only when it triggers movement or on a reset to rejoin.
 - AU915
 - Hotspots (names) used to transfer data -> unsure the openlns gwmp doesnt give us the hotspot name, b58 or asserted coords.

*NOTE: do not post your device `app_key`*

**Additional context**
the workings seem to be intermittent and it does seem to interfere with moving the device to another LNS and route id.
sorry the screenshots are not perfect. the fdn console devices have been disabled for last week or more and were added to our LNS route id one joined after some time and the other was just sending join/accepts.

i have released both these device euis off our route and put them back onto the fdn console for time being.  

**Screenshots**
If applicable, add screenshots (from Console) to help explain your problem.
[image]
[image]

macpie, 2023-07-11:
Thank you I really appreciate the details, we will bellowing into this.

michaeldjeffrey, 2023-08-01:
I believe this should be fixed by PR https://github.com/helium/router/pull/985.

Inactive devices were being incorrectly filtered out of the local cache of devices, so were not being considered for removal when we compared them to the list of devices from Console.

closed 2023-08-01 by michaeldjeffrey