helium/router #993: Add option to charge for late packet

macpie, 2023-08-15:


closed 2023-08-16 by macpie

NRC1978, 2023-08-16:
Really?

macpie, 2023-08-16:
> Really?

Yes, we have added the **options** for Router operators to charge for late packets. This is due to the fact that now (post [HIP-70](https://github.com/helium/HIP/blob/main/0070-scaling-helium.md)) late packets are charged to the OUI owner so we wanted to make sure that the cost can be passed down to the end user if needed.

ccall48, 2023-08-22:
I'm not really interested in buying late packets, if they dont arrive on time thats the networks problem. Why should i get charged for late packets?

michaeldjeffrey, 2023-08-22:
> I'm not really interested in buying late packets, if they dont arrive on time thats the networks problem. Why should i get charged for late packets?

By the time the packets make it to Router, the OUI owner has already paid for them. As in the above comment, an **option** has been added for a Console/Router LNS operator to pass the cost they've already payed to the device that incurred it.

ccall48, 2023-08-22:
Its not really an option for openlns on which is shoe horned around this. the packets incurred is really just a trust me bro tbh.

michaeldjeffrey, 2023-08-22:
> Its not really an option for openlns on which is shoe horned around this.

I'm not sure what you mean by this.

> the packets incurred is really just a trust me bro tbh.

Also this. A late packet is being verified to a specific devices set of keys. It's not a guess.

ccall48, 2023-08-22:
well of course, we need to wrap the functionality of external lns's around the console way of doing things.

there is no way for us to test that the packets are real or not is all i'm saying, so it is a trust me bro. i put one device doing uplinks every 10 minutes when we first connected our OUI and go charged for 10k packets in a few days its not a hill i'm willing to die on as the cost in minimal. 

is there any way this can be audited from a oui operators end on what device and packets sent so we can check the validity?

michaeldjeffrey, 2023-08-22:
> well of course, we need to wrap the functionality of external lns's around the console way of doing things.

Again, I'm not sure what you mean by this. External LNS are free to operate however they please. They're free to charge DC to orgs/tenants/what-have-you however they please. Or have no mention of DC at all.

> there is no way for us to test that the packets are real or not is all i'm saying, so it is a trust me bro.

In this situation, a packet is "real" if it can be properly decoded by a device's keys. Unless you're referring to a different sort of "real".

> is there any way this can be audited from a oui operators end on what device and packets sent so we can check the validity?

The event will go to the device that the packet was validated for. And if needed we can help a Router operator dump those packets for further inspection.

---

I'm wondering if there's some confusion between Router and Packet-Router.
This change is only for Router. By the time a packet has made it to Router, the OUI has already paid for it. An option is being provided so they can decide wether they want to eat that cost for their users, or pass it on the org that is receiving the late packets. It has no effect on Packet-Router. It has no effect on an LNS that is not Console/Router with this option disabled.

ccall48, 2023-08-23:
> I'm wondering if there's some confusion between Router and Packet-Router.
This change is only for Router. By the time a packet has made it to Router, the OUI has already paid for it. An option is being provided so they can decide wether they want to eat that cost for their users, or pass it on the org that is receiving the late packets. It has no effect on Packet-Router. It has no effect on an LNS that is not Console/Router with this option disabled.

It is possible, most of the names of the moving parts are very similar in this project and makes it hard to work out which piece fits where in the eco system.